chvmp/champ #107: Changing Gazebo Physics

HB-Liang, 2022-11-15:
Hi there, recently I keep moving on quadruped and solve that annoying initial position problem.
I come up with another problem that I need a quadruped working with a drone in a same gazebo world.
But they are not compatible as my drone works fine in the world physics with max_time_step=0.004 and real_time_update_rate=250.
In the same world, I use spawn_robot.launch in champ_config as a default spawning.
And here is what it behaves:
[image]
Would you please give some advise on changing quadruped setup?
BTW, quadruped works fine in a world with physics max_time_step=0.001 and real_time_update_rate=1000, but drone doesn't.
Thanks.

grassjelly, 2022-11-16:
I’m afraid you can’t run 2 different physics parameters in a single Gazebo world. You might need to decide which robot to adapt to another

HB-Liang, 2022-11-16:
That's the point, would you please give me some advise on adjusting quadruped's parameter to run in world with 0.004&250?

RamCharanThota, 2022-11-18:
Hi @HB-Liang, I am not sure this answer might help you to solve your issue. I had same issues with robot  jumping around. For me, it is mainly due to PID values in the controls configuration values of the robot. I have tuned them until stabilising the robot at start. Hopefully, it will resolve your issue.

grassjelly, 2022-11-29:
closing this as the issue is more Gazebo related. Feel free to re-open this

closed 2022-11-29 by grassjelly