chvmp/champ #111: ROS2 humble load_controller error when launch gazebo.launch.py

Ericsii, 2023-01-01:
Error when launch `gazebo.launch.py`

```
[ros2-9] usage: ros2 control load_controller [-h] [--spin-time SPIN_TIME] [-s]
[ros2-9]                                     [--set-state {configured,active}]
[ros2-9]                                     [-c CONTROLLER_MANAGER]
[ros2-9]                                     [--include-hidden-nodes]
[ros2-9]                                     controller_name
[ros2-9] ros2 control load_controller: error: argument --set-state: invalid choice: 'start' (choose from 'configured', 'active')
[spawn_entity.py-8] [INFO] [1672562455.472806756] [spawn_entity]: Spawn Entity started
[spawn_entity.py-8] [INFO] [1672562455.473142149] [spawn_entity]: Loading entity published on topic /robot_description
[spawn_entity.py-8] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-8]   warnings.warn(
[spawn_entity.py-8] [INFO] [1672562455.475532359] [spawn_entity]: Waiting for entity xml on /robot_description
[spawn_entity.py-8] [INFO] [1672562455.478280339] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-8] [INFO] [1672562455.478715671] [spawn_entity]: Waiting for service /spawn_entity
[ERROR] [ros2-9]: process has died [pid 26831, exit code 2, cmd 'ros2 control load_controller --set-state start joint_states_controller'].
[spawn_entity.py-8] [INFO] [1672562456.232808071] [spawn_entity]: Calling service /spawn_entity
[spawn_entity.py-8] [INFO] [1672562456.300750106] [spawn_entity]: Spawn status: Successfully sent factory message to Gazebo.
[INFO] [spawn_entity.py-8]: process has finished cleanly [pid 26829]
```

Ericsii, 2023-01-01:
From the document for ros2_control, `set_state` can only be `configured`, `active`

https://control.ros.org/master/doc/ros2_control/ros2controlcli/doc/userdoc.html#load-controller

grassjelly, 2023-01-01:
hi @Ericsii are you launching it directly or from a config package?

Ericsii, 2023-01-02:
I just launched it directly. However, I think this error is just due to the CLI argument `set_state` being changed in `ros2_control`.

Ericsii, 2023-01-02:
It doesn't seem to affect normal function in the simulation, just an error

Ericsii, 2023-01-02:
I solve this error by updating this line to `cmd=['ros2', 'control', 'load_controller', '--set-state', 'activte', `

https://github.com/chvmp/champ/blob/88c3e71f3a6678e6edce02e77b6a8ce7c74f9212/champ_gazebo/launch/gazebo.launch.py#L132

Tiryoh, 2023-01-08:
I think that's the part I forgot to fix in my pull request #109 . :cry:
https://github.com/chvmp/champ/pull/109/files#diff-c5d4e542a85ef638ec11e688a2338ffad349a9f34e52c9279962e03e4fa5c70a

Tiryoh, 2023-01-08:
Changing champ/champ_gazebo/launch/gazebo.launch.py like this seems reasonable.

```diff
    load_joint_state_controller = ExecuteProcess(
-       cmd=['ros2', 'control', 'load_controller', '--set-state', 'start',
+       cmd=['ros2', 'control', 'load_controller', '--set-state', 'active',
             'joint_states_controller'],
        output='screen',
    )
```

grassjelly, 2023-01-09:
hey @Tiryoh . Do you mind sending a PR, please?

Tiryoh, 2023-01-09:
@grassjelly Sure!

Tiryoh, 2023-01-09:
https://github.com/chvmp/champ/pull/112
Could you have a look, @grassjelly?

grassjelly, 2023-01-09:
thanks @Ericsii . Fixed here: https://github.com/chvmp/champ/pull/112 . Closing this for now. Thanks @Tiryoh for the PR

closed 2023-01-09 by grassjelly