chvmp/champ #116: Issue with the robot walking

RamCharanThota, 2023-02-06:
Hi,
We are testing our robot walking with champ. Robot is drifting sideways when walking forward.But, its walks relatively well when walking backwards. We are using PID as low level control of motor joints. We tuned the parameters several times. But, it didn't improved the forward walking performance. I really appreciate if anyone one can help with this issue.

chrisalbertson, 2023-02-07:
First thing to do with a new robot is verifying the kinematics is correct.   Command one foot to move off the ground 50 mm then measure if it is exactly 50 mm then command it to move 50 mm fore and aft and measure.  Measuring is hard and you might need to make tools if you want sub-millimeter accuracy.   Move each joint to "zero radians" and verify the angle.  Again you may need to print some tools.

Almost always you find the as-built robot is does not exactly match the software's internal model.  For example, gear back lash or the rotation sensor is not is not mounted exactly right

Why it walks differently forward of back is likely fore/aft weight distribution.  Or if the knees point with the direction of movent or away from it.  I bet the legs bend under load a little and do it differently

RamCharanThota, 2023-02-18:
Thank you @chrisalbertson  for the suggestions. I will try them.

vignesh-robotics, 2024-06-14:
Hi @chrisalbertson, I am facing the same issue, I am just launching in rviz visualization and getting the issue.
[champ_issue.webm](https://github.com/chvmp/champ/assets/100715955/a5deb620-d488-4539-8368-5d5551af8b95)

vigneshThangavelK, 2024-06-18:
@grassjelly, Hi can you give me any idea to check this issue. I can try to fix it.