chvmp/champ #133: Where could I find initial body height ?

elpimous, 2023-09-21:
Hi friend, where could I find, in code, the body Z value on initial pose ? 
When we run bringup, robot is up on its legs.
I'd like it to be sitted on ground (for a correct standup, with specific kp and kd)
Thanks @grassjelly 
Vincent

grassjelly, 2023-09-21:
you can add world_init_z here https://github.com/chvmp/champ/blob/master/champ_config/launch/gazebo.launch#L31

elpimous, 2023-09-21:
Hey, @grassjelly happy to hear from you..
no does t work.
Well how to have a robot sitted when first launch ? 
I have this:

And i'd like to have this, then build a standup function

elpimous, 2023-10-14:
@grassjelly Hello, any up ? 
Need to start controller with a sitted down starting pose ( in real robot)
Your above command doesn't produce any different result