chvmp/champ #134: Hardware Interface read() write()

elpimous, 2023-10-14:
Hello @grassjelly .

in your Hardware Integration page, you mention : **Publish all the actuators' current angle using [sensor_msgs/JointState](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/JointState.html) to 'joint_states' topic.**

However, it is not used into your code.

1/ Do you have anywhere, a hardware interface with both read and write functions ?  
2/ Could i read imu values, to stabilize robot walk (any idea where i could start ? My imu is working and correctly publishing.)
Take care of you, Juan.
Vincent

grassjelly, 2023-10-15:
hey Vincent, the joint_states is mainly for autonomy (odometry calculation).

You can use an IMU's reading to control the body pose [here](https://github.com/chvmp/champ/blob/master/champ_base/src/quadruped_controller.cpp#L98). Here's a pseudo code.

```
auto feedback = readIMU();
auto target_pose = req_pose_;
auto new_pose = pid(target_pose, feedback);
body_controller_.poseCommand(target_foot_positions, new_pose);
leg_controller_.velocityCommand(target_foot_positions, req_vel_, rosTimeToChampTime(ros::Time::now()));
kinematics_.inverse(target_joint_positions, target_foot_positions);
```

elpimous, 2023-10-15:
Hey, Juan. Thanks for feedback