chvmp/champ #145: Traversing the staircases

SourabhPrasad, 2024-06-29:
Hi, 
I am using the package with the provided spot_config. My goal is to enable the robot to autonomously navigate aws_hospital_word. However, the robot encounters difficulty in traversing the staircases within the world.

Please advise on how to configure the package to traverse the stairs.
Thank you!

https://github.com/chvmp/champ/assets/20325484/9bd99cb0-9b5a-47e3-92f4-2a634503c515

chrisalbertson, 2024-07-01:
I don’t think this is a configuration issue,  It is more of a “you have to write the software for stairs yourself” kind of issue.

One of the issues is that the robot’s standing pose (that is the pose before you consider leg motion) has to move the robot's center of gravity “up slope”, not just for stairs but any non-level surface.

With stairs you also have to change the way the gait in generated, the parameters used are not what you want, you need to think about exact locations for each foothold, and you can no longer just think in terms of stride length.     I think stairs are a little like walking across a creek using stepping stones.  Each foot has to land on a stair tread.

Then there is the anti-collision problem.  The steps might occupy the space a leg needs to swing through.  Remember too that steps might have “nosing” or a slight overhang.   The robot needs to keep a model of the stairs for anti-collision.

This means you need to add a feature because as far as I see it, chvmp/champ does not consider that a foot might contact the ground during a swing.  It would not be good at stepping over rocks on a dirt path.  I think stairs are like this.

I don’t have any code to share, I have only started to figure out what’s needed.   I want the robot to be able to walk on rough outdoor paths including stairs.   I working on a real robot and have a poorly working prototype and the next two revisions in the pipeline.

I’d be interreted to hear what you have figured out so far.

> On Jun 28, 2024, at 6:04 PM, Sourabh Prasad ***@***.***> wrote:
> Hi,
> I am using the package with the provided spot_config. My goal is to enable the robot to autonomously navigate aws_hospital_word. However, the robot encounters difficulty in traversing the staircases within the world.

SourabhPrasad, 2024-07-05:
Hey @chrisalbertson!,
Thank you for the reply.
I am also in the early stages of development, trying to figure out how to proceed. 
I am currently looking into possibly integrating these packages ([Quad-SDK](https://github.com/robomechanics/quad-sdk) & [elevation_mapping_cupy](https://github.com/leggedrobotics/elevation_mapping_cupy))

SourabhPrasad, 2024-07-11:
@chrisalbertson
Check out [this](https://github.com/asblab-research/tbai). I have been testing the controller in it and they work well.