chvmp/champ #150: Help usign champ with hyperdog quadruped robot

diego112891, 2024-11-22:
I am working on a project focused on mapping and navigation using the HyperDog quadruped robot in ros2 (https://github.com/NDHANA94/hyperdog_ros2). This is the final goal of my project.
However, I have encountered issues with the HyperDog robot  due to a bug that causes it to only slide instead of walking. The creator of the robot mentioned that they have not been able to find a solution for this issue. To address this, I plan to implement the CHAMP control framework, as the discussions in the HyperDog repository suggest it is compatible for controlling the robot.

Before implementing CHAMP in ROS2, I wanted to test it in ROS1 to use the Robot Setup Assistant. However, I am facing difficulties in doing so. Specifically, I would like to ask for your guidance on the proper way to create the URDF. I generated a URDF from the xacro files, but the body of the robot does not appear in Gazebo, nor can I control it co
[image]
rrectly. In rviz (the whole robot appears correctly but I cant moveit or make it stand up . 
[image]
[hyperdog_completo.txt](https://github.com/user-attachments/files/17875700/hyperdog_completo.txt)
https://espolec-my.sharepoint.com/:u:/r/personal/diadrodr_espol_edu_ec/Documents/Datos%20adjuntos/[champ_prueba_Ws](https://espolec-my.sharepoint.com/:u:/r/personal/diadrodr_espol_edu_ec/Documents/Datos%20adjuntos/champ_prueba_Ws.zip?csf=1&web=1&e=uzQh90).zip?csf=1&web=1&e=uzQh90

diego112891, 2024-11-26:
update

the body of the robot finally appeared in Gzebo the problem was that I left a space " " in the name of one the links and i also replaced the line  `<mesh filename= "File:///home/diego/champ_prueba_Ws/src/urdf_hyperdog/meshes/body.stl" scale="0.001 0.001 0.001"/>` with `<mesh filename="package://urdf_hyperdog/meshes/body.stl" scale="0.001 0.001 0.001"/>` and that resolved the issue.

the bad news is that the robot still doesnt work properly. When te controllers are loaded instead of standing up this happens and also doesnt move

[image]

and this is what happens when i try to move it rviz
[image]

i dont know what to do to make the robot work properly 
maybe this part of the urdf is not well declared?

[image]

or maybe the problem in the  gait parameters?

[image]

davidromeroy, 2024-12-03:
Help to my friend @grassjelly