chvmp/champ #155: Default robot stance too narrow

luisemilio14, 2025-05-07:
Greetings, 
I am trying to configure a custom robot (ANYmal D) using the champ framework. I am using the ANYmal C base parameters for gait, links and so on, and put the ANYmal D urdf on top of it. After modifying the PID parameters a little bit, I could get an initial controller that manages to make the robot walk for some time.

My problem is that the robot quickly loses its balance when walking. I suspect this is due to the default stance being too narrow, resulting in a small support polygon.

[image]

I would like to know if there is any straightforward way of configuring this parameter,. I checked the gait configuration and experimented a bit with it, but to no avail.

Thanks in advance.

chrisalbertson, 2025-05-07:
> luisemilio14
>  created an issue

Stance width is a kind of double edged sword.   Wider gives a larger polygon when standing on all four legs, but as soon as the robot walks, the “polygon” becomes a “line" with zero width.   Assume the case of walking straight forward.    The robot translates forward relative to the suport “line”.  And the center of gravity moves left or right relative top the line.     A smaller stance with means there is less leteral movment of the CG relative to the line.

There is an optimal stance wiidth and it seems to be to chance.  You want it to be wide when stopped of moving VERY slow and narrow as speed picks up.

I have a real dog and I watch her in a Trott, she keeps her legs apart only enough so the feet don't collide.  This has the effect of making the “support line” nearly parallel with the direction of movement.   This minimizes lateral balance shift durring the stride.

When she goes even faster she switches to a gallop where the front and back legs move together.    In this case the “suport line” is at 90 degrees to the motion and there is zero lateral balance shift.

I tried to get my robot to use a wide stance and it requires some odd sideways leg motion to keeep the CG over the line, Narow stances don’require this.

In short, as soon as the robot lifts two legs off the ground, the polygon vanishes and you have zero static stability.  The ‘bot becomes an inverted pendulum.

luisemilio14, 2025-05-15:
Thanks for the insight! Do you have any tips on how I can modify the robot parameters in order to get a more stable locomotion pattern? I really hope it does not come down to trial-and-error tuning of the PID parameters...

LucasC19, 2025-07-18:
Thanks for the reply.

Regarding the controller I had the opposite situation. I started setting the controller for position and the robot was "exploding" in simulation since the joints were moving instantly to the requested position. The problem was solved using effot controller. 

About the stance width I was able to solve the problem. Apparently CHAMP, in the default stand position, tries to align the Upper Leg Link and The Foot Link. My problem was that in the URDF both links were displaced. Once I aligned them vertically, the problem was solved.